BelaPlatform/bela-hardware #15: bela_cape-eagle-import.kicad_sym not found

birdshao, 2024-02-25:
when open the bela_cape_rev_v3 project using kicad, bela_cape-eagle-import.kicad_sym not found, where can get it?

mxa, 2024-02-27:
Just checking, we're talking about `bela_cape_rev_C3`?

birdshao, 2024-02-27:
> Just checking, we're talking about `bela_cape_rev_C3`?

Yes, when load the project bela_cape_rev_C3 to kicad, it seems missing some footprint library and package library

mxa, 2024-02-27:
.kicad_sym is a symbol in the schematic, not a footprint. It's just referenced in the file `sym-lib-table` but not actually used. Delete the file `sym-lib-table` and it should open without the error message.

giuliomoro, 2024-02-27:
@mxa does this need `sym-lib-table` shouldn't be included in version control?

mxa, 2024-02-27:
I've removed it with commit 555166d entirely. If it has needed paths to local libraries it should be included in the version control, but in our case it contained just one reference to a long deleted and obsolete local library.

birdshao, 2024-02-27:
> `.kicad_sym` is a symbol in the schematic, not a footprint. It's only referenced in the file `sym-lib-table` but not actually used in the schematic. Delete the file `sym-lib-table` and the project should open without the error message.

There's still some errors. I import footprint library bela.pretty and Bela.kicad_sym(where I found in the folder .\belamini\belamini_cape_rev_C1) to kicad. When I update footprints and symbols from library, the error shows like:
Error: Updated footprint IC6 (Bela:QFN40P500X500X80-41N-D): *** library footprint not found ***
Error: Updated footprint G1 (Bela:Bela-Logo_30mm_Silkscreen): *** library footprint not found ***
Error: Updated footprint P1 (Bela:BeagleBone_Black_Sockets_Board_outline): *** library footprint not found ***
and
Error: Update symbol P1 from 'Bela:BeagleBone_Black_Header' to 'Bela:BeagleBone_Black_Header': *** symbol not found ***
Error: Update symbol G1 from 'Bela:Bela-Logo' to 'Bela:Bela-Logo': *** symbol not found ***
Error: Update symbol P1 from 'Bela:BeagleBone_Black_Header' to 'Bela:BeagleBone_Black_Header': *** symbol not found ***
Error: Update symbol IC6 from 'Bela:ES9080Q' to 'Bela:ES9080Q': *** symbol not found ***
Error: Update symbol IC1 from 'Bela:PAM8406' to 'Bela:PAM8406': *** symbol not found ***
(If not import Bela.kicad_sym, there will be 3 more errors)

So I guess some footprints and symbols are missing in the libraries. I can download the footprints and symbols of some  missing ICs from mouser, but I can't get the exact library version of some ICs like BeagleBone_Black_Header\Bela-Logo which matches the project.

birdshao, 2024-02-27:
> @birdshao I don't think any of this is really an issue since KiCad stores copies of all used symbols and footprints in the schematic and layout file anyways. So you can open these files and look at the layout for debugging or even manufacture the PCB if you like. In any case, I've added symbols and more footprints with commit [5fe71ca](https://github.com/BelaPlatform/bela-hardware/commit/5fe71ca13c240c204a8c8fe817fc99a9a0d7c39a)

Yes, U are right. Thanks for your patient help. I'm not familiar with kicad. I find out I can open the symbol or footprint which is missing in the library first, then save and add it into the library. By that way, the library will be complete.